bigtreetech/BIGTREETECH-TouchScreenFirmware #2921: Marlin TFT Screen Glitch

nichols89ben, 2024-03-29:
### Description
Started getting a screen glitch at random moment, but happens every print at some point. Ive searched forums but havent found a solution that works. I also tried the section that was in troubleshooting

#define ST7920_DELAY_1 DELAY_NS(200) // After CLK LOW
#define ST7920_DELAY_2 DELAY_NS(400) // After DAT
#define ST7920_DELAY_3 DELAY_NS(200) // After CLK HIGH
### Steps to reproduce

### Hardware Variant
TFT70

### TFT Firmware Version & Main Board Firmware details
bugfix-2.1.x 2024-03-25
CONFIG_VERSION 20230929 

### Additional Information
[Archive 2.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/14807577/Archive.2.zip)

[image]
[image]

radek8, 2024-04-04:
A similar problem has been reported here in the past.
My display did it a few times too. But this problem has not been reported for a long time. Maybe also because people are buying touchscreens so they don't have to use Marlin mode :-)
Are you sure you are using the latest FW for TFT?

radek8, 2024-04-04:
The solution could be shielding the cabling leading to the display.
see this post:
https://github.com/bigtreetech/BTT-TFT35-E3-V3.0/issues/80#issuecomment-1762893295

closed 2024-04-11 by nichols89ben

radek8, 2024-04-11:
Have you found a solution to your problem?

nichols89ben, 2024-04-12:
> Have you found a solution to your problem?

Sorry, I added some of the adjustments suggested in the Marlin firmware and I bought some shielding. It hasn't happened again yet so one of those fixed it. Now I just need to figure out how to get idex options to show in the BTT touchscreen ui!