bigtreetech/BIGTREETECH-TouchScreenFirmware #2927: Filament sensor detect filament runout

Blisk, 2024-04-10:
### Description

When printing after a 30 minutes I get filament runout error and pausing print. Even there is no filament runout.
### Steps to reproduce

I push Icon resume and than printing continue for some time and than again filament runout.

**Expected behavior**
I get filament runout only when there is no filament.

**Actual behavior**

### Hardware Variant
TFT35 and SKR mini 3.0 and smart filament sensor

### TFT Firmware Version & Main Board Firmware details
marlin 2.0.x

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.
[image]
[image]
[image]

rlneumiller, 2024-04-12:
To eliminate the possibility of this being a problem with the filament runout sensor itself, connect the runout sensor wires together (this will result in the sensor reporting that the filament is always present). If the issue no longer occurs then it is a hardware issue; likely a low quality runout sensor microswitch or wire\connector failure.

MrKuskov, 2024-04-15:
If you are using the latest version of the TFT35 firmware, set up communication with Marlin https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2910

Blisk, 2024-04-16:
I have tested filament sensor and it works ok.
communication is not a problem

rlneumiller, 2024-04-16:
When did this issue start occuring? What changed?
If you can zip and attach your configuration.h, it might help with running this down.

rondlh, 2024-04-19:
How is the filament sensor wired? To the main board or the TFT? Can you provide the exact details?

Blisk, 2024-04-20:
to the TFT display

burneme, 2024-05-21:
Hola, I had in the past same problem, the limitation for **fil_runout_distance** to 50mm wasn't useable as you can easily see in my setup below.
The unnecessarily heavy SMART_FILAMENT_DETECTOR is steady at my machine (of course not mounted to the print head), so it pulls out sometimes more when going to the corner and doesn't pull for a while when printing in the middle.
Anyhow I had to change the firmware to increase the 50mm barrier. Still got sometimes the same error as you and looked deeper in. The routine for the smart filament detection wasn't to understand, so I wrote my own changes and now it's working fine for me. Even in the front panel I display the amount of filament before **runout** will occur. If you are able to compile the sw by your own, I can give the changes to you. It concerns mostly **AddonHardware.c**.
This is a Ender-5plus with MS Direct drive and 7inch display.
Regards

[image]

LinoFG, 2025-01-11:
Hi,
[burneme](https://github.com/burneme)
What are the changes that solve the issues in the AddonHardware.c file?