bigtreetech/BIGTREETECH-TouchScreenFirmware #2928: [Q] It stops during printing.

vfbank, 2024-04-13:
I am using Marlin 2.1 bugfix version.
If the printer's speed exceeds 100 mm, the G code of the SD card and USB port of the touch screen cannot be read properly and printing stops.

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware
The settings at the address above have been completed. The firmware is also up to date.

For reference, there is no problem with Btt_mini 12864 v2.0.

I tried the communication speed from the default setting to 100000, but it is the same.

What I found most strange was that it worked normally in 12864 emulation mode rather than graphics mode.

Does anyone know anything about this issue?

MrKuskov, 2024-04-15:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2910

vfbank, 2024-04-15:
> #2910

Thank you for your assistance. I have implemented the settings you suggested, but I'm still experiencing a bug where certain G-codes are not read correctly, and the machine moves to an incorrect axis before returning. Since it works fine with the mainboard's built-in TF card, it doesn't seem to be an issue related to buffer or communication speed.

kisslorand, 2024-04-15:
@vfbank
You might want to try the FW from [my repository](https://github.com/kisslorand/BTT-TFT-FW).

P.S.:
Some trolls might come and bark here how dangerous that FW might be for your printer, but you can ignore them, they are just upset that I didn't show them the source code for my FW that made it superior to the one from this repository.
Haters gonna hate....

vfbank, 2024-04-16:
> [내 저장소](https://github.com/kisslorand/BTT-TFT-FW)에서 FW를 사용해 볼 수 있습니다.
> 
> 추신 : 일부 트롤은 FW가 프린터에 얼마나 위험할 수 있는지 여기에서 짖을 수 있지만 무시할 수 있습니다. 싫어하는 사람은 싫어할 것입니다.

Thank you for your help. I've tried applying the firmware mentioned, but printing still stops when the speed increases. I'm using Simplify3D and Orca slicer. Both show the same issue: when exceeding 100mm/s (with acceleration over 3000), printing either stops or there's a phenomenon where it moves to x or y during printing and then returns.

I've also tested with other BTT displays. All TFT-type displays show the same symptoms. I've tried wrapping the cables with shielding foil in case of noise issues, but it didn't work.

What drives me even more crazy is that excluding this TFT display, everything works perfectly fine on both mainboards (SKR mini and SKR3 EZ mainboard with EZ2209 stepper drivers). It prints flawlessly from fast speeds up to over 300mm/s. Only this TFT series functions properly at speeds below 100mm/s.

closed 2024-04-16 by vfbank

vfbank, 2024-04-16:
It has been confirmed that the issue is caused by insufficient performance of the hardware chipset. So I gave up on this product and the marlin and switched to Klipper.

I would like to thank everyone who helped me. I will close this thread.

karabas2011, 2024-06-07:
"there's a phenomenon where it moves to x or y during printing and then returns."
I noticed the same on difference versions of marlin earlier with common display reprap discount smart controller and different boards. Suddenly X moves to X0 then returns and go on