bigtreetech/BIGTREETECH-TouchScreenFirmware #2929: MKS TFT28 Using "Master" - "Copy to SD Card root directory to update" not working in English

SpankyDevil, 2024-05-08:
MKS TFT28
Firmware MKS_TFT28_V3.0.27.x.bin
config.ini
And tried all the TFT28 Themes

The result is the firmware install is in Chinese so I can't understand it but once completed, the dashboard works correctly but is in Chinese.

I have tried adding the en language file, i have also tried setting language settings to 1 to use another language in this case English but with no success.

rondlh, 2024-05-09:
**My recommendations:**

1. Download the latest build: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/archive/refs/heads/master.zip
2. Inside the archive select the firmware you need (MKS_TFT28_V3.0.27.x.bin)
3. Select the language pack you want (language_en.ini).
4. The default language is English as defined in config.ini (should be: "language:0")
5. Flash these 3 files, and you should be greeted in English.

SpankyDevil, 2024-05-12:
> **My recommendations:**
>
> 1. Download the latest build: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/archive/refs/heads/master.zip
> 2. Inside the archive select the firmware you need [...]

Its like the language files are corrupted, I tried different ones and none are the language they are meant to be.

SpankyDevil, 2024-05-16:
[image]
[image]
[image]

SpankyDevil, 2024-05-16:
I deleted the word_unicode.fon file from within the TFT28\Font\ Folder and created a blank file with the same name.
Reflashed and now everything is working in English.
[image]
[image]
[image]

closed 2024-07-24 by github-actions[bot]