bigtreetech/BIGTREETECH-TouchScreenFirmware #2933: [Q] Bad serial connection, advanced ok gives checksums, without printer freezes. Next steps?

Norvat, 2024-05-20:
I am running a TFT 35 E3.0 using firmware BIGTREE_GD_TFT35_V3.0_E3.27.x (Pre-Compiled)

Mainboard SKR 1.3 with tmc 2209 steppers in UART running marlin 2.1.2.2

I cannot get a stable connection between the tft and mainboard. See pictures bellow for errors. 

What i have tried so far:
- Replaced the tx and rx wires going to the tft with a shielded and grounded one
- Updates both marlin and tft to latest releases
- Tested all different baudrates from 9600 to 1 000 000
- Changed the serial ports in firmware
- Tried turning on advanced ok and checksum, and the printer does work, but it gives out errors constantly. Sometimes lines of gcode is lost and the printer skips a line. Without this feature the print freezes, but can still be controlled from the tft. While frozed "pending gcodes" is 1, but nothing is transmitted. 

[image]

[image]

[image]
[image]
[image]
[image]

It seems it has some serious character loss

All dependencies in marlin is included, and all features in the tft work as they should. 

Zip of marlin if that is helpfull: 
[Marlin-2.1.2.2.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/15377280/Marlin-2.1.2.2.zip)

Running the printer from pronterface works and does not give errors. 

I am currently at a loss about what i should try next to mitigate the problem, any help is greatly appreciated

ChihebMadiouni, 2024-05-20:
we have the same problem with MKS TFT35 , MKS ROBIN Nano V3 on 25 machines 
i have tested all the baudrates and ports same issue 
the print always stops mid job , when printing from the TFTSD or the TFTUSB 
we are using the marlin 2.1x bugfix with `advanced_ok` active ,

Norvat, 2024-05-20:
@ChihebMadiouni Hi, its good to know im not allone with these issued. What troubleshooting steps have you tried?

kisslorand, 2024-05-20:
@Norvat @ChihebMadiouni 
Guys, I highly recommend try the firmware from [my repository](https://github.com/kisslorand/BTT-TFT-FW).

Norvat, 2024-05-20:
@kisslorand running a test print now, will report back

digant73, 2024-05-20:
if you don't have a reliable connection, simply disable advanced_ok and command_checksum.

Norvat, 2024-05-20:
@digant73 Disabling those features causes the printer to freeze. Do you have any ideas how i could troubleshoot?

Norvat, 2024-05-20:
@kisslorand Your firmware returned a "unknown command" error 30 min in, but a lot better than how it was running with the stock firmware. Any tips to how i could troubleshoot the serial connection? I have made my own shielded cable so the cable should be good. 

[image]

Baudrate is currently set at 57600

ChihebMadiouni, 2024-05-20:
i tested a 1 hour print with @kisslorand FW version with 250k baudrate and the print finished no errors and no stops 
however i will continue testing to be sure

Norvat, 2024-05-20:
@ChihebMadiouni Do you think running a lower baudrate is causing my problems? I set it lower during testing to make sure i wasn't overloading the tft or mainboard

ChihebMadiouni, 2024-05-20:
@Norvat 
i recommend you switch to marlin 2.1.x bugfix the release 2.1.2.2 may have some bugs 
also some features i have enabled in marlin maybe it will help 
`#define ADVANCED_OK`
`#define TX_BUFFER_SIZE 32`
`#define RX_BUFFER_SIZE 1024`
`#define MAX_CMD_SIZE 96`
`#define BUFSIZE 4`
`#define SERIAL_OVERRUN_PROTECTION`
`#define SERIAL_DMA`
`#define EMERGENCY_PARSER`

Norvat, 2024-05-20:
Thank you so much for suggestions 

I will investigate further later this week

digant73, 2024-05-21:
> @digant73 Disabling those features causes the printer to freeze. Do you have any ideas how i could troubleshoot?

The print freezes when the TFT doesn't receive an ACK from the mainboard (e.g. due to EMI). When the TFT has no more available TX slot (1 in case advanced_ok is disabled) (it means also the TFT has pending commands) then the TFT will not send any command

MrKuskov, 2024-05-23:
@Norvat у меня были те же ошибки на B3 
вот мои настройки. Все работает корректно.
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2910#issuecomment-1988557561

Norvat, 2024-05-23:
Did some more testing. 

Tried to print without the heated bed, and printed for an hour without errors. I then set bed temp, and the printer ether stopped right away or did some weird moments that isn't in the gcode. It seems to be especially bad while heating up. 

Its a 24V 500x500mm bed (Probably pulling 15A), and the controller is placed about 5cm away from it. The original controller board was in a metal enclosure but i have made a plastic one to fit the SKR board, so i think the issue is EMI from the heated bed. I tried to pack the electronics enclosure it in aluminum foil that was grounded, but it did not work. My wrapping might have been a bit shoddy. 

I am going to try to move the electronics away from the printer next.

kisslorand, 2024-05-23:
Try what happens if you put a fairly big capacitor on the wires of the bed heater, something like 4700uF~10000uF / 36V(or higher).

rondlh, 2024-05-26:
@Norvat Please be careful with Kisslorand's closed source firmware. It's known to be buggy, unreliable and slow. His FW is far behind what you can find here. There is a reason why his "contributions" are ignored https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pulls

Norvat, 2024-05-26:
@rondlh Thank you for the helpful comment. I will do some measurements after i get access to my oscilloscope again in a week or two. 

The psu is the original that came from tronxy (this is a tronxy x5sa 500 pro), so it might not be the best. I have a smaller spare psu, so i will try to run power to the bed separately from the controller. 

If the psu is the issue, would it be due to a varying voltage or some frequency?

rondlh, 2024-05-27:
> If the psu is the issue, would it be due to a varying voltage or some frequency?

That's difficult to predict. To me it's clear that there is something seriously going wrong. The PSU could be noisy, and/or insert high frequency noise or even dip down to a low voltage for a short time when overloaded. If the voltage would dip, then the MCU on the motherboard and TFT would probably freeze, so that's not very likely, but a scope will show you.

You could actually do some test without heating the bed, that might help to point you in the right direction. I recommend to leave the TFT checksum feature on, so you will be informed if any serial corruption takes place.

Another thing you could try is to add an external serial connection to listen to the serial data flow, make sure to only listen (RX of your added port, don't connect TX). On your computer you then can use Putty to see the data flow. So you connect your external RX to the RX or TX on your motherboard. If you connect it to RX you can see the data the motherboard receives from the TFT, if you connect it to TX you can see the data the motherboard send to the TFT.

digant73 has recently added the "info screen" that can be help diagnose the problem. He sent me a few lines of code to diagnose my serial communication issues in the past. Insert just before "loopProcess();" in Monitoring.c, for testing go to the info screen. You can uncomment one of the 3 mustStoreCmd lines (below the 1st one is uncommented).
Please remove this code when running actual prints, because if you would accidentally go to the info screen it will cause issues during a print!
The code makes the TFT send commands to the motherboard as fast a possible to which the motherboard responds. If everything is going right you should just see lots RX/TX commands and data in the info screen, the numbers should be relatively stable. If they go to 0 then the serial communication has broken down.

```
  static uint32_t counter = 0;
  if (!isFullCmdQueue()) {
    sprintf(str, "M118 TEST %d\n", counter++ % 512); mustStoreCmd(str);
  //  mustStoreCmd("M118 P0 A1 test test test test test test test test test test test test test test test test test\n");
  //  mustStoreCmd("M43\n");
  }
```

Norvat, 2024-06-05:
Small update:

I have a separate PSU to the controller, and that seems to have done the trick with the errors, and the printer runs happily.

BUT! 

If i ground the cable from the controller to the tft, i get errors. I can start a print without it connected and the second i connect it i get errors. 

I tried scoping both the ground on the original psu and i am getting ripples of +-3V. I see +-0.5V through the shielding on the tft cable while it is not grounded. Almost no ripple from the new PSU. I tried getting pictures of the readings but the oscilloscope is an old analog one, so not too easy. 

So i think the issue is psu related as rondlh says

Will test a bit more scientifically later this week, and try the tips above

karabas2011, 2024-06-07:
MKS Eagle marlin 2.0.9.3 and 2.1.2.2+ MKS35 TFT with last btt dev firmware.

Printer stops suddenly.  Not hanging. With or without encanced_ok / crc.
Tried  all. 
Finally tried kisslorand's firmware - Success of14 hours print.

karabas2011, 2024-06-11:
I flashed Marlin 2.1.2.3 yesterday + last BTT fw.  Shortened cable to 20cm. Nothing changes. Sudden stop.

rondlh, 2024-06-18:
@karabas2011 Please be careful with Kisslorand's closed source firmware. It's known to be buggy, unreliable and slow. His FW is far behind what you can find here. There is a reason why his "contributions" are ignored https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pulls. Kisslorand's closed source FW might damage your motherboard and/or your TFT display, so better stay clear.
Also, the topic you raise here seems to be unrelated to the issue raised by Norvat, please start a new issue if you need support.

karabas2011, 2024-08-19:
> @karabas2011 Please be careful with Kisslorand's closed source firmware. 
His  pull files is open source. I applied it manually (relative to serial problems).  Yes it is not for last version. But works for low speed. Currently no other solution found. Wierd. I think this serial problem  is marlin internal problem. I faced with it even with PC repetier-host - marlin based board connections (different boards). Unpredictable always.

closed 2024-10-27 by github-actions[bot]