bigtreetech/BIGTREETECH-TouchScreenFirmware #2938: Installed MicroProbeV2, but can't use BLtouch panel?

AbyssNullImage, 2024-05-30:
So I just recently installed a BIQU MicroProbe V2
Double checked that pins match with that on board
Followed directions from instructions provided
Previously, I did have a BL touch that worked fine, until it snagged a corner of a print and destroyed the probe pin
After that, I put in the new probe but in Touch mode, the BL touch panel no longer works
When starting machine, probe goes through self test cycle and clears, but when entering BL Touch panel
Throws a "Unknown command: "M280 P0 S90" and then just hangs there
I keep pressing okay, but the only way to get out is by restarting the machine
I even tried through the terminal, inputting command M280 P0 S10
Spits out as Unknown command as well
Then after that, it repeats "P31 B3" and just keeps repeating "M155 S3" below it and it loops, but I can back out of the terminal
Weird thing is, if I go through the Marlin mode side of things, it'll deploy and stow the probe without issue, even self test
Same thing in Touch mode, if I go through Home, it'll complete all endstops, X, Y, Z and even clear Auto Z Tilt
Even doing UBL, it'll do as it's suppose to
So can someone explain why I'm not able to use it? I just want to have in it in case I want to test the probe for functionality.

Hardware:
BTT SKR 3 EZ [Board]
BigTreeTech_GD_TFT50 [Screen]

Firmware:
Marlin_Bugfix-2.1.x
BigTree_GD_TFT50_V3.0.27.x

Norvat, 2024-05-30:
The BlTouch and MicroProbe have different configuration in marlin, have you reconfigured and flashed new firmware? 

MicroProbe uses #FIX_MOUNTED_PROBE instead of #define BLTOUCH

Details can be found here: [https://github.com/bigtreetech/MicroProbe/blob/master/MicroProbe%20V2%20User%20Manual_20240330.pdf](url)

AbyssNullImage, 2024-05-30:
Yes, it works in the Marlin mode. Even Touch modeas well.
Main issue is not being able to deploy or stow in Touch mode, but all fine in Marlin mode.
As for configuration, there was a sub section for #define MicroProbeV2, which I had enabled

closed 2024-08-06 by github-actions[bot]