bigtreetech/BIGTREETECH-TouchScreenFirmware #2942: Tool Change option for T0 and T1

Manojmonotech, 2024-06-21:
Hi 
I would like to create a new icon for Tool change option in BTT TFT50. V3. I created the icon but it is not showing in display. 
Can anyone help me to solve this.   I need the carriage should move left when i click T0 and right when i click T1.

rondlh, 2024-06-21:
You can use the "Custom menu" for this, or just type T0/T1 into the Terminal screen.

Manojmonotech, 2024-06-22:
Hi Rondlh
Thanks for your reply. I would like to keep this icon in main screen and when i click T0 carriage should move left and move right when i click T1. I created the icon. But don't know how to give command for this icons. 
Can you please help

rondlh, 2024-06-22:
What do you mean by this?
> when i click T0 carriage should move left and move right when i click T1
Do you mean that the carriage should park? I assume it's an IDEX system, right?
The behavior/movements for T0 and T1 are defined by the host software, which I assume is Marlin.
What screen do you call the "main screen"? The screen immediately after power on? Which icon positions have you chose for T0 and T1?

Manojmonotech, 2024-06-22:
Hi 
As per your advice i edited in the display config custom menu and tested. it is working ok. Thanks for your help. 
I need the tool change option  in the TFT main menu. I have created 2 BMP file for this purpose.  
After adding new BMP file it doesn't show in the touch screen. Any possibility to make it work please ?

rondlh, 2024-06-23:
That seems correct, are you looking in the right place?

Manojmonotech, 2024-06-24:
Hi
I am not getting the right way to link the BMP file.

closed 2024-08-31 by github-actions[bot]