bigtreetech/BIGTREETECH-TouchScreenFirmware #2943: Hide the BED Icon

eerthanaK, 2024-06-24:
Since I am using a chamber and not using a bed, I have disabled the bed temperature. Additionally, I need to remove or hide the bed icon from the BIGTREETECH TFT50 V3 display.

I am using Marlin software on the mainboard.

rondlh, 2024-07-01:
That's good to know, thank you for the information.

zonalimitatore, 2024-07-11:
Where you have disabled the HB? Just in Marlin config.h or also in TFT firmware (config.ini)?

#### Heated Bed Support
#   Options: [disable: 0, enable: 1]
heated_bed:0

eerthanaK, 2024-07-15:
> Where you have disabled the HB? Just in Marlin config.h or also in TFT firmware (config.ini)?
> 
> Heated Bed Support Options: [disable: 0, enable: 1] heated_bed:0

I disabled the heated bed in the marlin config. h and TFT firmware. But even then It is flashing from the heated bed and chamber in the display status screen

eerthanaK, 2024-09-16:
Good Morning Team,
                                Already, I do the heated_bed: 0 is disabled
in touch screen display. status screen is enabled but the chamber and bed
icon is showing alternatively. If the status screen is disabled in chamber
Icon only shown.

Regards & Thanks
Keerthana T

NOVAXIM, 2024-10-14:
I will join this postulate.
I do not have a heated bed, but when I choose "Preheat" I have (Heat) "Both" and "Hotbed", instead of limiting the options to "Nozzle" only. On status screen "Bed" the tile is present but it is ok.
Of course, I have disabled hotbed in Marlin and TFT configurations.

closed 2024-12-22 by github-actions[bot]