bigtreetech/BIGTREETECH-TouchScreenFirmware #2944: [Q] USB Print doesn't start LCD Print

felipebraz, 2024-06-27:
I have an SKR E3 Mini V3 + TFT35 with Marlin 2.1.2.3 and everything is working fine except when I starting printing from USB (Octoprint).
When I start printing from USB the LCD doesn't open the printing screen, it keeps an state as iddle. The print occurs and everything works except this part.
How solve this issue?

kisslorand, 2024-06-27:
EGR Off

rondlh, 2024-07-09:
So you use Octaprint to start a print job that is located in the USB port of the TFT?
How are things connected together?

Are you sure you are starting the job in the USB port of the TFT?

I don't use Octaprint, I use ESP3D, which is connected to the 2nd TFT serial port, from there I can start a print job located in the USB or SD port of the TFT. That works fine.

felipebraz, 2024-07-09:
No, the Octoprint is connect via USB directly to the SKR Board.

I found a workaround to make it work sending the updates do LCD through GCODE, but it's not very reliable.

rondlh, 2024-07-09:
So you are printing from Octaprint? If so, what do you want the TFT to do in that case?

felipebraz, 2024-07-09:
Monitor and follow the printing status, I don't want to open an app or PC just to see the progress and other status if I have an LCD waiting for me.

rondlh, 2024-07-09:
In your way of printing, the TFT doesn't know what's going on, it's not part of the Octaprint to motherboard serial communication. The TFT only receives very limited information like temperature reports and perhaps some fan speed, printer feed rate and flow info. It will not receive any progress or expected print time information that can be displayed.

Typically the print is run from the TFT, then the TFT is in control and you can get a status overview.

zonalimitatore, 2024-07-11:
If you start to print from an external program/terminal/sender, how can display full interact with that program? 
You'll be able to control flow/speed/temp/light/printer settings (acceleration, save babystepping values)/maybe fan % (but if the gcode "recall" at every layer the fan speed, you have to continue to change that). 
I mean, you can control almost all, but you haven't the "print screen as print from sd/usb"

rondlh, 2024-07-11:
> If you start to print from an external program/terminal/sender, how can display full interact with that program?

The display cannot, it doesn't have access to the gcode file that is being printed.

>You'll be able to control flow/speed/temp/light/printer settings (acceleration, save babystepping values)/maybe fan % (but if the gcode "recall" at every layer the fan speed, you have to continue to change that).

Marlin will accept commands from defined serial sources, so you can still control your printer though the TFT, even during printing. Other serial sources can overrule your gcode commands, commands are processed in the order they are received by Marlin.

What concrete information are you missing? Print progress?

felipebraz, 2024-07-11:
> If you start to print from an external program/terminal/sender, how can display full interact with that program? You'll be able to control flow/speed/temp/light/printer settings (acceleration, save [...]

In fact you can have, there ia trick you can do in GCODE, I found an workaround that works almost fine in issues is this repo.
About the display recognizes the print, it can. Other printers with OEM LCD display works just fine. Even my previous display (Generic DWIN) with custom Marlin worked just fine. It recognizes when a print starts from an external event, show the updates, you can interact with the printer, you can even cancel the print on Octoprint directly on the display.
Marlin sends a code for the display informing there is print starting, status, movements,... In this case (TFT35) the display is not "hearing" it.
Look at this issue: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2631

zonalimitatore, 2024-07-11:
> The display cannot, it doesn't have access to the gcode file that is being printed.
>
>
> Marlin will accept commands from defined serial sources, so you can still control your printer though the TFT, [...]

We're saying the same thing, man.

felipebraz, 2024-07-11:
> > If you start to print from an external program/terminal/sender, how can display full interact with that program?
> 
> The display cannot, it doesn't have access to the gcode file that is being printed.

Yes, the display can know, as it seems the display firmware has issues in it's codding.
Someone already debbuged and found some inconsistencies (https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2631#issuecomment-1328135446).

> Marlin will accept commands from defined serial sources, so you can still control your printer though the TFT, even during printing. Other serial sources can overrule your gcode commands, commands [...]

Yes, it is the main information that is missing.

felipebraz, 2024-07-11:
> > > If you start to print from an external program/terminal/sender, how can display full interact with that program? You'll be able to control flow/speed/temp/light/printer settings (acceleration, save babystepping values)/maybe fan % (but if the gcode "recall" at every layer the fan speed, you have to continue to change that). I mean, you can control almost all, but you haven't the "print screen as print from sd/usb"
> > 
> > 
> > In fact you can have, there ia trick you can do in GCODE, I found an workaround that works almost fine in issues is this repo. About the display recognizes the print, it can. Other printers with OEM LCD display works just fine. Even my previous display (Generic DWIN) with custom Marlin worked just fine. It recognizes when a print starts from an external event, show the updates, you can interact with the printer, you can even cancel the print on Octoprint directly on the display. Marlin sends a code for the display informing there is print starting, status, movements,... In this case (TFT35) the display is not "hearing" it. Look at this issue: #2631
> 
> Is your "serial always on"=1 ?

I honestly don't know... LoL
I'm learning about this display as I talk with you. I have it for just 2 weeks. Previously I used to have another solution.
But I'll search about it.

rondlh, 2024-07-11:
If you use the display in Marlin mode, then Marlin is in full control of the display, and the progress can be shown. If you use the display in serial mode (which I have been assuming up till now) then things are not so easy because even Marlin doesn't know the progress, unless the gcode commands tell Marlin (M73). Marlin can be configured to send a progress report (M73_REPORT), which could be picked up by the display. This is currently not implemented.

felipebraz, 2024-07-11:
> If you use the display in Marlin mode, then Marlin is in full control of the display, and the progress can be shown. If you use the display in serial mode (which I have been assuming up till now) [...]

Yes, I believe it is the case. Will be great if implemented some day.

rondlh, 2024-07-12:
I just came across this section in Mainboard_AckHandler.c

```
  else if (ack_seen(":print_start"))  // print started from remote host (e.g. OctoPrint etc.)
  {
    startPrintingFromRemoteHost(NULL);  // start print originated and hosted by remote host and open Printing menu
  }

```

Can someone check the terminal when an OctaPrint print job is started, to see what is received by the display?

closed 2024-09-18 by github-actions[bot]