bigtreetech/BIGTREETECH-TouchScreenFirmware #2949: [BUG] (short description) TFT35 display no longer updates temperatures from printer

dmarnel, 2024-08-20:
### Description

TFT35 display no longer updates the hot end or bed temperatures from the printer until the print begins.

Machine: Creality Ender 3 Pro

Printer board: BTT SKR Mini E3 V2.0
	chip: STM32F103 RCT6

Display board: BTT TFT35 E3 V3.0
	chip: STM32F207 VCT6

Marlin bugfix-2.1.x (as of 8-19-2024)

BIGTREE_TFT35_V3.0_E3.27.x.bin (as of 8-19-2024)

### Steps to reproduce

1. Initiate print

**Expected behavior**

Hot end and bed temperatures should update on the display while the printer is heating up.

**Actual behavior**

Hot end and bed temperatures do not change until the print begins.

### Hardware Variant

TFT35 E3 V3.0
### TFT Firmware Version & Main Board Firmware details

BIGTREE_TFT35_V3.0_E3.27.x.bin (as of 8-19-2024)
Marlin bugfix-2.1.x (as of 8-19-2024)

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.
[Configuration.h.zip](https://github.com/user-attachments/files/16676586/Configuration.h.zip)

radek8, 2024-08-30:
Set Parameter
emulated_m109_m190:1

dmarnel, 2024-08-30:
Thanks, but that is already set.

radek8, 2024-08-30:
How old is your FW in the display?

radek8, 2024-08-30:
Send both Marlin configuration files and ini TFT file

dmarnel, 2024-08-30:
Thank you, I Appreciate the help. Here are the files.

[Archive.zip](https://github.com/user-attachments/files/16821213/Archive.zip)

radek8, 2024-08-30:
The configuration is fine.
The display should update the temperature as soon as you start printing.

radek8, 2024-08-30:
In the display menu, do you have the M109, M190 (green dot) emulation option enabled in the settings?

dmarnel, 2024-08-31:
Yes, M109 is enabled. Here are my settings:

[image]
[image]
[image]
[image]
[image]

closed 2024-11-08 by github-actions[bot]

MontanaCA, 2024-12-04:
Has a fix for this issue been identified?

dmarnel, 2024-12-05:
Not that I am aware of, but I haven’t updated the firmware since then.