bigtreetech/BIGTREETECH-TouchScreenFirmware #2950: [BUG] (short description)

Blisk, 2024-08-26:
### Description

TFT display shows some strange errors and prints are always failed.

### Steps to reproduce

printing from SD card
printing from octoprint

**Expected behavior**
Print should be printed ok but it is not and errors should not show up.

**Actual behavior**
I get this errors and failed print. Always.

### Hardware Variant
TFT 35 3.0
### TFT Firmware Version & Main Board Firmware details

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.
[image]
[image]
[image]
[image]
[image]

radek8, 2024-08-30:
According to the errors, it can be seen that you have a problem with the serial communication between the display and the printer.
Try reducing the transmitted speed.
Use a shielded serial cable between the display and the printer!
Keep the cable as short as possible and try to keep as much distance as possible between the cable and the stepper motor drivers and also between the serial cable and the cables to the motors.
The motor drivers generate interference that oscillates on the serial line.

radek8, 2024-08-30:
The original cable supplied with the display is not shielded and is not suitable in case of major interference.

Blisk, 2024-09-02:
> The original cable supplied with the display is not shielded and is not suitable in case of major interference.

thank you. Where I can get this cable which is shielded??? I was searching on aliexpress and didn't find that kind of cable. Do you know where I can buy it?

radek8, 2024-09-02:
I made it myself

Blisk, 2024-09-02:
> I made it myself
> 
> [#1837 (comment)](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1837#issuecomment-852237045)

did you have the same problems as I did before you did that?

radek8, 2024-09-02:
Very randomly

radek8, 2024-09-02:
It seems that SKR min E3 V3.0 boards suffer greatly from this problem. A lot of people with this board have this problem.

Fortunately, I have another board :-)

radek8, 2024-09-02:
I had a different problem. I connected the Wifi module and when printing it sometimes happened that the antivirus in my PC tested the serial ports and sent a query to the printer that blocked my printing. That's why I only connect the wifi module when I need to tune some parameters.

After using a shielded cable I solved the random print stop due to a corrupted command like in your case.

Try that shielded cable and possibly different speeds on the serial port.
You can also try disabling or enabling OK Advance. maybe some of it will take. You can also try shielding the cable to the stepper motors.
Your problem is that the serial port of the printer where the TFT is connected is oscillating some kind of interference that damages the communication between the printer and the display. The printer then cannot process such a corrupted command and returns an unknown command.

Blisk, 2024-09-02:
Thank you for your advice.
I don't have enabled "advance" in TFT display it is OFF. I will made my own shielded cable and see what is going on. I have connected to OctoPrint and testing that.

Blisk, 2024-09-02:
> I made it myself
> 
> [#1837 (comment)](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1837#issuecomment-852237045)

Is a connector to connect TFT display this one ?
JST XH2.54

radek8, 2024-09-02:
make XH2.45 on both ends

Dioniusos, 2024-10-21:
Got  same  uknown commands,   (have 2  enders3 SKR E3  with TFT GD35)  did  shielded  cable on one but I did not help , lowered  speed to 57600, did not help either,  compiled  latest 2.1.2.4 bugfix  that did trick, does not  display  uknown command  anymore.

Blisk, 2024-10-21:
> Got same uknown commands, (have 2 enders3 SKR E3 with TFT GD35) did shielded cable on one but I did not help , lowered speed to 57600, did not help either, compiled latest 2.1.2.4 bugfix that did trick, does not display uknown command anymore.

NO shielded cable didn't help. Will try to compile latest. Did you do  2.1.2.4 bugfix for TFT display or motherboard SKR mini

Dioniusos, 2024-10-21:
I did for   SKR mini,   had   older  2.1.2   that had no issues with, (no  uknown commands at all ever) but   i upgraded  extruder and  had to recompile   2.1.2.4 to change direction (was too lazy to  rewire cable) , i belive it was bugfix that had DMA2  for TFT fix - pr #27385  and lower  speed to 57600  just in case. ( just lowering  speed did not help, so I guess it was  latest bugfix that helped.)

Blisk, 2024-10-22:
> I did for SKR mini, had older 2.1.2 that had no issues with, (no uknown commands at all ever) but i upgraded extruder and had to recompile 2.1.2.4 to change direction (was too lazy to rewire cable) , i belive it was bugfix that had DMA2 for TFT fix - pr #27385 and I lower speed to 57600 just in case. ( just lowering speed on older firmware did not help, so I guess it was latest bugfix that helped.) Previously I used Marlin 2.1.2 on mobo and screen with that TFT firmware: https://github.com/kisslorand/BTT-TFT-FW on 2 ender3 clones with same motherboards and same TFT screens for a year and issues started after I 've upgraded to 2.1.2.4 both machines, so I figure it out that there is no issue with serial cable interference, rather TFT or SKR firmware.

thank you for info I will do this now and see. Because now I have also a problem with Z seam it makes a bump on one side even if there is no Z seam. And starting vircle is smooth but end is really rough. Banging my head why and still didn't figure out. But only with circles.

Dioniusos, 2024-10-26:
I have GD TFT  version , so used  [ this](https://github.com/kisslorand/BTT-TFT-FW/blob/main/Copy%20to%20SD%20Card%20root%20directory%20to%20update/BIGTREE_GD_TFT35_V3.0_E3.27.x.bin) 
( Accidentally once I flashed  non  GD version by  mistake - use it for months till I realize it was wrong  version. Actually did not noticed any  issues even was not GV version.
Does the bumb happen alwaes at the same hight ?   
I usually troubleshoot by  eliminating all posibilities, like corrupted slicer issue, ( been there -  random bumps happened only on   specific prints)  corrupted model, Use other slicer jusdt to  see if it makes any diffrence, use simmilar model,  then i 
check for  pom wheels  wear off, or most offen replace them,  I do check belt  tension,  I never had  z banding isssues since i did use oldham type couplings.

Blisk, 2024-12-12:
random bumps on circled models are because of Y axis linear rails, I tried twice and every time when I mount on Y axis I get bad prints. Now I use wheels as it was

digant73, 2025-02-07:
those unknown command messages are probably fixed by PR #2970. Increase the value for new param `tx_delay` (you can change it also at runtime from the Feature menu) until the issue is no more present.

closed 2025-04-30 by bigtreetech