bigtreetech/BIGTREETECH-TouchScreenFirmware #2952: MKS TFT35 v1.0 firmware update stuck on Updating Configuration

nisi1t, 2024-08-31:
I have a MKS TFT35 version 1.0 display, i try to update the firmware MKS_TFT35_V1_0.27.x.bin

- the card contain file MKS_TFT35_V1_0.27.x.bin -> renamed as MKSTFT35.bin
- config.ini
- TFT35 folder (from THEME_Unified Menu Material theme)

after inserting card firmware start to update, pass 100% , start to upload fonts and images, and on the step Updating Configuration
it load 130 keywords and stop there. 

Powering down after 30 min i see that MKSTFT35.bin renamed into MKSTFT35.cur - i understand that firmware is processed, the TFT35 folder and config.ini is unchanged

I did tried with a config.ini unchanged, i verified to be saved with UTF-8 encoding, formatting SDCARD and tried again

Nothing help to finish the firmware update,

Is something else i can do ?

radek8, 2024-09-02:
Try another SD card

nisi1t, 2024-09-13:
reseting or changing card did not work, the fix was to compile the firmware and from code debugging i notice that if i ad a small delay when the config is written on EEPROM the settings are saved , without the delay the code hang on EEPROM update

closed 2024-11-20 by github-actions[bot]

digant73, 2025-02-09:
@nisi1t which delay value and where did you put it in the code? so I will put it in the next TFT release