bigtreetech/BIGTREETECH-TouchScreenFirmware #2953: Updated to firmware to Marlin 2.1 and now TFT 35 e3 is not working

Byrnes9, 2024-09-02:
I tried updating my firmware today to Marlin Bugfix 2.1 with an SKR Mini e3 v2.0 and TFT35 e3 V3.0. My  printer was working fine but now nothing will appear on the tft35 e3 screen in Marlin mode and "Printer not attached", appears in Touchscreen mode. I've tried updating the firmware of both the screen and printer to no avail. This was working before I tried to update.

rondlh, 2024-09-04:
> I tried updating my firmware today to Marlin Bugfix 2.1 with an SKR Mini e3 v2.0 and TFT35 e3 V3.0. My printer was working fine but now nothing will appear on the tft35 e3 screen in Marlin mode and [...]

If the display says "Printer not Attached", it is probably because the baud rate of the display and Marlin are not the same. Make sure that the serial port in Marlin that connects to the display is enabled and works at the same  baud rate as the display. You can change the baud rate in Marlin's Configuration.h, and on the display.

For the display to work in Marlin mode, you must define the display in Marlin's Configuration.h, enable REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

radek8, 2024-09-04:
> > For the display to work in Marlin mode, you must define the display in Marlin's Configuration.h, enable REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

Attention.
Byrnes9 writes that he has a SKR mini E3 V2.0
This board has only one EXP connector.
Therefore, in Marlin, you must define the CR10_STOCKDISPLAY display.

Display controller REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER must remain disabled.

Byrnes9, 2024-09-04:
> Attention. Byrnes9 writes that he has a SKR mini E3 V2.0 This board has only one EXP connector. Therefore, in Marlin, you must define the CR10_STOCKDISPLAY display.
>
> Display controller [...]

This is how I have it set with nothing coming up on either Marlin or touchscreen mode.

Baud rate is correct.

radek8, 2024-09-04:
send configuration files

radek8, 2024-09-04:
Did both modes work before the update?

Byrnes9, 2024-09-04:
> send configuration files

[Files.zip](https://github.com/user-attachments/files/16880927/Files.zip)

I dont have my pc at the moment so I’ll send the full thing once I do

Byrnes9, 2024-09-04:
> Did both modes work before the update?

yes, I did notice a new file in Marlin called Config.ini so I do wonder if that could be the cause.

radek8, 2024-09-04:
Also send the config.ini you uploaded to TFT

Byrnes9, 2024-09-04:
> Also send the config.ini you uploaded to TFT

 I used the one on the BTT GitHub. https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

radek8, 2024-09-04:
> > Also send the config.ini you uploaded to TFT
> 
> I used the one on the BTT GitHub. https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware
> 
> I also used the non-GD firmware

Yes, you used the one on the website, but you probably modified it for your printer before using it. Or not?

radek8, 2024-09-04:
Port 2 is fine.
The display driver for the Marlin is also fine.

You should enable the following values ​​in the settings:

define M114_DETAIL  
define REPORT_FAN_CHANGE
define AUTO_REPORT_POSITION
define M115_GEOMETRY_REPORT
define HOST_ACTION_COMMANDS
define SERIAL_FLOAT_PRECISION 4

Byrnes9, 2024-09-04:
> > > Also send the config.ini you uploaded to TFT
> > 
> > 
> > I used the one on the BTT GitHub. https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware
> > I also used the non-GD firmware
> 
> Yes, you used the one on the website, but you probably modified it for your printer before using it. Or not?

I didn’t change any settings here.

Byrnes9, 2024-09-04:
> Port 2 is fine. The display driver for the Marlin is also fine.
> 
> You should enable the following values ​​in the settings:
> 
> define M114_DETAIL define REPORT_FAN_CHANGE define AUTO_REPORT_POSITION define M115_GEOMETRY_REPORT define HOST_ACTION_COMMANDS define SERIAL_FLOAT_PRECISION 4
> 
> Read the instructions on the splash screen.

I’ll change them once I get a chance and will report back.

Byrnes9, 2024-09-06:
> Port 2 is fine. The display driver for the Marlin is also fine.
>
> You should enable the following values ​​in the settings:
>
> define M114_DETAIL define REPORT_FAN_CHANGE define AUTO_REPORT_POSITION [...]

I've made these changes but no improvement, both tft and marlin mode won't work

Byrnes9, 2024-09-06:
I found the solution, it ended up being the fault of the SD card, a replacement solved it. Closing the issue now.

closed 2024-09-06 by Byrnes9

Kandrey2024, 2024-09-07:
Пожалуста помогите. Решил прошить TFT35 E3 V3.0 Но получил керпич. Я в этом не чего не понимаю. Нужна прошивка для него.

Kandrey2024, 2024-09-07:
Found outdated data:

Fonts: Update required
Config: ok
Language: Update required(Optional)
Icons: Update required
Insert the SD card with required
files and press the reset button to update

Вот это на экране. И что делать не знаю.