bigtreetech/BIGTREETECH-TouchScreenFirmware #2956: [FR] Biqu Microprobe support

salembegic, 2024-09-08:
The BTT TFTs have a manu for manually testing the BlTouch probe. In there you can test the probe, deploy it, stow it..., but there is currently no support for the Biqu Microprobe. Since the Microprobe works based on the pin state and does not emulate the servo motor, current buttons just show an error when pressed, as the G code commands are not recognised.

I would like to have an option to choose between the Bltouch probe and Microprobe in configuration.ini, and if the microprobe is selected it should show a different menu with buttons that emulate the appropriate comands.

rondlh, 2024-09-27:
Can you provide more technical information or technical reference for the Microprobe?

salembegic, 2024-09-27:
Hi, yes, here is the github page: https://github.com/bigtreetech/MicroProbe. Deploying/stowing seems to be done by changing the state of the control pin, and the reading of the state is the same as with BLtouch

rondlh, 2024-09-27:
I read the document, here is the key related info.

There are 2 versions, V1 and V2. V1 triggers on high, V2 triggers on low.
5 pin interface, in same order as BL/3D Touch (Signal - Ground - Control - Vcc (5V) - Ground)
The probe needs 5V power (max 350mA).
Compatible with 3.3V and 5V control signal.
Signal is a open-drain output which requires a 10K pull-up resistor. V2 requires a "strong pull-up" (value is not specified, 10k ohm?), 
The control pin is not a serve pin(like with BL 3D Touch), it just sets the deploy/stow state. High is Deploy, Low is Stow. The deploy time is 500ms.

A 10K ohm pull-up cannot be achieved with the MCU internal pull-ups. I believe the TFTs have no external pull-ups, so one might need to add an external pull-up for the probe to work (especially for the V2).

If you just want to do some testing then you can define custom commands in the TFT to deploy and stow the probe. The range of the probe seems to be very limited. The price is lower than the BL Touch, but much more expensive than the 3D Touch.

closed 2024-12-04 by github-actions[bot]