bigtreetech/BIGTREETECH-TouchScreenFirmware #2958: [BUG] Problem with reading files from SD card inserted to mainboard

NOVAXIM, 2024-10-12:
### Description

I use SD card inserted to mainboard, not touchscreen. I like use long name directories and files.

### Actual behavior

**Problem 1.**
When I have over any quantity files on card, then I can not reading card in touch screen mode. When I deleted some files, then I can reading the card. This problem not occurs on Marlin mode.

**Problem 2.**
In touch mode I try to start printing from file placed in directory I only hear short 3x BEEEP and it does not print. No other message.
I can start printing from this same file in directory without problems in Marlin mode.

### Hardware Variant
SKR MINI MZ V1.0 and TFT35-MZ V3.0

### TFT Firmware Version & Main Board Firmware details
SKR MINI MZ V1.0 with Marlin 2.1.2.4 (latest)
TFT35-MZ V3.0 with firmware [d6690b2a6d35315d635353ae7e48c51243ebe33c ](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/commit/d6690b2a6d35315d635353ae7e48c51243ebe33c)

### Additional Information
I tested renaming directories and files and this not resolve of problem with start printing.

closed 2024-12-22 by github-actions[bot]

digant73, 2025-02-09:
the TFT allocates memory for each file name and folder. In addition it store both the short name and the long name. So a lot of memory is used if there are a lot of file/folder in the same folder you are in that moment. So yes you have to limit the number of file/folder when reading the sd card plugged in the mainboard. The 3 beep you hear when you try to start a print are due to something wrong whit the path (the TFT requests to mainboard to print the file but the mainboard does not find the specified path), If 'm not wrong the max length for a file/folder is 63 chars. Try to not exceed that length

NOVAXIM, 2025-02-14:
Thank you for your explanation.
Before I had many more files/directories and had no problem. That's why I thought it might be caused by the new version. Maybe I was just impressed.

I have an additional question about this issue:
- is it possible to increase the TFT memory for reading from a card inserted into the motherboard?
- special characters (like ° or ®) use more memory?

digant73, 2025-02-17:
- the TFT try to allocate the memory needed until there is no more memory available. So there is no cap limit.
- I think it is not a special symbol making too much difference