bigtreetech/BIGTREETECH-TouchScreenFirmware #2959: [BUG] (short description)

7u5a, 2024-10-14:
### Description
When printing from the touch screen interface on the TFT35 E3 v3.0.1 2023.4.12, the target and actual temperature are not shown in the printing interface.

### Steps to reproduce
1. Select print from on board SD card
2. Select file to print

**Expected behavior**
Bed temp set point >0, bed starts heating with actual temp reported.
Hot end temp set point >0, hot end starts heating with actual temp reported.
Home, then print

**Actual behavior**
Bed and hot end temp set points stay at 0. Actual reported temp is same as when print was started.
Home, then print.
After 1st layer has started, temperatures are now reported correctly.

### Hardware Variant
BTT TFT35 E3 V3.0.1
BTT SKR Mini E3 V3.0

### TFT Firmware Version & Main Board Firmware details
Mainboard - Marlin 2.1.X bugfix (self configured)

### Additional Information
[MarlinConfigFiles.zip](https://github.com/user-attachments/files/17359143/MarlinConfigFiles.zip)

7u5a, 2024-10-15:
By the way, in Marlin mode it seems to work...

Dioniusos, 2024-10-21:
Have  same   behavior,   my hardware  BTT TFT35 E3 V3.0.1 GD
BTT SKR Mini E3 V3.0 and Marlin 2.1.2.4 bugfix

rondlh, 2024-10-21:
First of all, the TFT works best if you don't print from the onboard SD card because like this the TFT doesn't know well what's going on, it can only rely on the Marlin reported information.
If the TFT receives a valid temperature report, the temps (actual and target) will be updated and shown. This will always work, unless the TFT is "not connected" as reported on the display.
For the TFT to work best, you should enable a number of Marlin options, check the TFT "config.ini" file for more details.

In Marlin mode the TFT pixels are fully controlled by Marlin, so then the behavior is completely different.

closed 2024-12-29 by github-actions[bot]