bigtreetech/BIGTREETECH-TouchScreenFirmware #2960: TFT35 throws M14 E unknown command

Blisk, 2024-10-16:
### Description

When I print something I get error unknown command M14 E
### Steps to reproduce

Start printing over octoprint and get this error

Print withouth errors

### Hardware Variant

TFT35 V3
### TFT Firmware Version & Main Board Firmware details

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.
[image]
[image]

rondlh, 2024-10-21:
This is an error thrown by the motherboard host software, the display just reports it. 
I'm not sure what host software you are running, but Marlin doesn't understand M14, so it would report the error you got.
(https://marlinfw.org/docs/gcode/M010-M011.html)

Make sure to select the correct gcode dialect when slicing.

Blisk, 2024-10-21:
it is also Marlin 2.1.X on SKR mini 3.0

radek8, 2024-10-26:
This error is usually caused by electromagnetic interference on the serial port

Blisk, 2024-10-26:
> This error is usually caused by electromagnetic interference on the serial port. SKR mini V3 boards suffer a lot from this

I have now shielded sreial cable

radek8, 2024-10-26:
It doesn't always help. It looks like a bad design of the board, maybe the interference from the stepper motor controller is oscillating to the serial port. Put the cables as far away from the stepper motor controller as possible.

Blisk, 2024-10-26:
> It doesn't always help. It looks like a bad design of the board, maybe the interference from the stepper motor controller is oscillating to the serial port. Put the cables as far away from the stepper motor controller as possible.

OK I will do that also.

radek8, 2024-10-26:
A lot of people here complain about this problem and 90% of them have one thing in common - SKR mini V3.0

rondlh, 2024-11-01:
@Blisk Do you always get the "M14 E" error, or did it just happen once? If it only happened once then, like @radek8 mentioned, it could be a serial communication reliability issue. In that case lower your serial baud rate to 115200 baud, shorten your serial cable as much as possible and keep the serial cable away from the stepper cables.

Blisk, 2024-11-01:
> @Blisk Do you always get the "M14 E" error, or did it just happen once? If it only happened once then, like @radek8 mentioned, it could be a serial communication reliability issue. In that case [...]

yes that error happend many times and I have shielded serial cable.

radek8, 2024-11-01:
Is the error code still the same, or is it changing and is different each time? 
(text in quotation marks)

Blisk, 2024-12-12:
code is mostly the same

kisslorand, 2025-01-04:
@Blisk 

Test the FW from [my repo](https://github.com/kisslorand/BTT-TFT-FW). Some monkey loitering around (@rondlh) might tell you to beware because that FW might burn your house, empty your bank accounts, break your printer or any other stupidity he comes up with but you should be smart enough to think for yourself. 
Other people did the switch and never looked back.

Blisk, 2025-01-05:
Hi

I checked your FW but you only have TFT28 firmware i use TFT35 display.

What I need to do here?

digant73, 2025-02-07:
@Blisk that unknown command messages are probably fixed by PR #2970. Increase the value for new param `tx_delay` (you can change it also at runtime from the Feature menu) until the issue is no more present.

closed 2025-04-30 by bigtreetech