bigtreetech/BIGTREETECH-TouchScreenFirmware #2961: [BUG] TFT35 stops responding with red 'Busy processing please wait....'

Blisk, 2024-10-22:
### Description

Sometimes when I print something and than want to check or level or something I get this busy processing and screen frezze.
### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

not to frezze at all

**Actual behavior**

it frezze 

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.
[image]

Blisk, 2024-10-23:
> I just had a look at your video, a bit short, but at least it's clearly shows the issue, from the configuration you posted I can identify the issue, so please change the related setting and try again.

I have updated marlin now. Will see what will happend

Blisk, 2024-10-23:
just doesn't work, after ABL started in 20 seconds I get this error.
I get this everytime I want to print.
from Octoprint I get also this.
Recv:  T:169.66 /170.00 B:59.92 /60.00 @:22 B@:20
Recv: X:235.00 Y:235.00 Z:-0.82 E:-2.00 Count X:18800 Y:18800 Z:-328
Recv: echo:busy: processing
Recv:  T:169.92 /170.00 B:59.94 /60.00 @:16 B@:18
Recv: echo:busy: processing
Recv: X:235.00 Y:235.00 Z:11.18 E:-2.00 Count X:18800 Y:18800 Z:4472
Recv: echo:busy: processing
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 +0.543 +0.215 -0.115 -0.043 -0.202
Recv:  1 +0.623 +0.268 -0.075 -0.280 -0.187
Recv:  2 +0.685 +0.317 -0.048 -0.290 -0.107
Recv:  3 +0.635 +0.373 +0.113 -0.038 -0.275
Recv:  4 +0.765 +0.342 +0.053 -0.080 -0.175
Recv: 
Recv: X:235.00 Y:235.00 Z:11.18 E:-2.00 Count X:18800 Y:18800 Z:4472
Recv: ok
Send: N38 G29 P1 X0 Y0 W50 H20 C*18
Recv: Error:Bad Z value
[image]

Changing monitoring state from "Printing" to "Error"

rondlh, 2024-10-23:
> What settings do you mean? a power restore?

The settings you posted here...

Blisk, 2024-10-24:
> > What settings do you mean? a power restore?
> 
> The settings you posted here...

Ok I posted alot of them, can you please let me know exactly which one?

rondlh, 2024-10-24:
The settings you posted here... 
>Include a ZIP file containing your Configuration.h or use [Pastebin](https://pastebin.com/) and paste a link in this issue.

Blisk, 2024-10-24:
> The settings you posted here...
> 
> > Include a ZIP file containing your Configuration.h or use [Pastebin](https://pastebin.com/) and paste a link in this issue.

Ok so I need to change this? to disable it?
#define NOZZLE_TO_PROBE_OFFSET { -40, -10, -1.85 }

rondlh, 2024-10-24:
> Ok so I need to change this? to disable it?
> #define NOZZLE_TO_PROBE_OFFSET { -40, -10, -1.85 }

You should not disable it, just make sure the number are correct.

radek8, 2024-10-26:
> 
![20241023_133940](https://private-user-images.githubusercontent.com/2773050/379255264-e2142180-3a38-4a5b-99bb-152c1224a30f.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk5NzA1NDAsIm5iZiI6MTcyOTk3MDI0MCwicGF0aCI6Ii8yNzczMDUwLzM3OTI1NTI2NC1lMjE0MjE4MC0zYTM4LTRhNWItOTliYi0xNTJjMTIyNGEzMGYuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTAyNiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEwMjZUMTkxNzIwWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YmQzY2NjNTAxM2QzZTRlYTM4NGJiYTY5YmI4MTJiZWFlOGNhYjcxYmY2Zjc2NGJhOWQwMWE3NmQ4ZDc4Nzg5YiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.W0L5jDwwB_vdzco2inrFbd3veLwYzkFuslFdKbFtRUE)
> 

This looks like the thermal protection of the printer has been activated.
Have you set the correct parameters?
THERMAL_PROTECTION_HOTENDS
THERMAL_PROTECTION_BED
THERMAL_PROTECTION_CHAMBER
THERMAL_PROTECTION_COOLER

Blisk, 2024-11-05:
> THERMAL_PROTECTION

this is all default I didnt setup any of this.
[Marlin.zip](https://github.com/user-attachments/files/17630317/Marlin.zip)

closed 2025-01-12 by github-actions[bot]