bigtreetech/BIGTREETECH-TouchScreenFirmware #2962: [BUG] (short description) print stops midway after installing tft3.3 v3 on ender 3 pro with bit skr mini motherboard.

solccd, 2024-10-30:
### Description

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

rondlh, 2024-10-31:
Thanks for the detailed description, it's clear what is causing this, just check the docs

solccd, 2024-11-06:
hi I have check everything flashing and updating the touchscreen firmware but the issue is there what can I do?

MLiv79, 2025-01-02:
It's a common issue since a long time, i've the same problem with my skr 1.4t and TFT 7.0. All user here mocking Kisslorand user about his fork but is the only one WORKING. The master, whatever version, doesn't work if you print from tft sd or tft usb. You've to print from motherboard SD card to finish a print, elsewhere you only wast a lot of filament because the print will hang at a certain moment. Download the kisslorand version (it is old but at least working) https://github.com/kisslorand/BTT-TFT-FW

digant73, 2025-02-09:
@solccd that should be fixed by PR #2970. Once merged, simply set the new param `tx_delay` to 1, or 2 (increase it until the issue is fixed). Also enable command checksum (`command_checsum` set to 1 in `config.ini` file) so you will always be notified by any data corruption on mainboard. The issue should be related to inefficient/ not reliable IO reading on some mainboards. The TFT properly sends data when requested by mainboard but the mainboard has a very limited implementation of IO reading (polling in most mainboards) so the data are not properly parsed (some chars are lost etc.). The param `tx_delay` on TFT is a workaround for a weakness on mainboard

closed 2025-04-30 by bigtreetech