bigtreetech/BIGTREETECH-TouchScreenFirmware #2965: Filament Runout sensor

eerthanaK, 2024-12-03:
Dear Sir,

We are currently trying to integrate the Smart Filament Sensor V1 with the SKR 3 EZ board and have followed the necessary steps. We have also disabled the display firmware to streamline the setup. However, despite our efforts, the filament sensor is not functioning as expected.

Below are the key points of our configuration:

The Smart Filament Sensor V1 has been physically connected to the SKR 3 EZ board as per the provided documentation.
We have updated the Configuration.h file with the relevant settings. Screenshots of the configuration are attached for your reference.
The filament runout sensor is enabled (FILAMENT_RUNOUT_SENSOR), with additional parameters configured, including FIL_RUNOUT_STATE, FILAMENT_RUNOUT_SCRIPT, and FILAMENT_RUNOUT_DISTANCE_MM.
We used internal pull-up resistors (FIL_RUNOUT_PULLUP) for the sensor pins.
Despite these configurations, the sensor does not seem to trigger any filament runout response during operation.

Could you please assist us by providing:

Guidance on whether the firmware settings are correctly configured.
Any additional steps or configurations required for compatibility with the SKR 3 EZ board.
Suggestions on troubleshooting the issue effectively.
Attached are screenshots of the Configuration.h settings for your review. If you require more details about the hardware or logs, please let me know, and I will be happy to provide them.

Thank you for your time and support. We look forward to your assistance in resolving this issue.
[image]
[image]

rondlh, 2024-12-05:
So which TFT are you using? And how is the runout sensor wired?

eerthanaK, 2024-12-18:
> So which TFT are you using? And how is the runout sensor wired?

We are using BIGTREETECH TFT50 V3.0