bigtreetech/BIGTREETECH-TouchScreenFirmware #2969: [Q] how to get Skr3 ez to work with TFT35

inspectionsbybob, 2025-01-29:
I am working on putting a SKR3/TFT35 into my Tronxy X5SA.

My printer is LARGE and the distance between the SKR3 and where I want the TFT35 is over a meter…

I was hoping to use use the 5pin cable only as I have an extension cable for that.

What do I need to setup in config.h for the SKR3 so that I can use marlin mode only or do I have things mixed up and the 5pin is for the touch mode?

I have not flashed the tft35 as I though the default software would work in marlin mode…

Again, my big issue is I only want to run -1- cable front to back, in whatever mode that is.  I use the Usb port and pronto for 99% of stuff (repetier actually)

Anyway, what do I need to do… I can set things up on the bench with the 3x factory cables, it seems that this combination is persnickity and not as easy as I thought it would be…

rondlh, 2025-02-01:
You are correct, the 5 pin cable is for the touch mode of the display, it establishes a serial connection.

In config.ini set default_mode to 1 or 3 (1 is the default value):

#### Default Mode
# Set Marlin/Touch Mode as the default mode at startup.
#
# NOTE: Mode switching is possible only for Marlin Mode and Touch Mode by a long press of
#       1.5 seconds on the display or holding down the encoder button for 1.5 seconds.
#
#   Options: [Marlin Mode: 0, Touch Mode: 1, Blocked Marlin Mode: 2, Blocked Touch Mode: 3]
default_mode:1

In Marlin you can disable the "Full Graphic Smart Controller".

I just added SERIAL_DMA to Marlin for the STM32H7, which will be a useful feature you could enable to improve serial communication reliability.

inspectionsbybob, 2025-02-01:
First: welcome back from Holliday.  Hope you had a good time.

I figured Ihad some sort of issue with my serial port setup.
There are 3 default ports that can be enabled.  I had them setup as -1, 0
and 1 with all at the default rate, but enabled the Mcode to change it.

I will, re-compile and verify the serial settings.

Rebuilding a system from scratch, doing both hardware and firmware at the
same time is challenging to say the least… never sure what the issue is
until you get one side stable.

-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

rondlh, 2025-02-02:
Thanks, I'm not sure if I fully understand. Anyway...
The serial port in Marlin works like this:
-1 is the virtual USB serial port
For the SKR 3 you should define serial port 1. There is no serial port 0. 
So you could set:
#define SERIAL_PORT     1
#define SERIAL_PORT_2   -1
 
The TFT will be connected to serial port 1.
Just match the serial baudrates of Marlin and the TFT, and you should be ready to go.

inspectionsbybob, 2025-02-02:
> <bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2969/2629081047@
> github.com>

Ok, I recompiled with the serial ports set differently and got enough error
and warning mesgs for it to fail.

Here are the error msgs and a few lines on either side

Marlin\src\HAL\STM32\../../inc/../core/serial.h: In function 'void
SERIAL_CHAR(char, Args ...)':
Marlin\src\HAL\STM32\../../inc/../core/serial.h:132:55: error: request for
member 'write' in 'multiSerial', which is of non-class type 'SerialOutputT'
{aka 'int'}

compilation terminated due to -fmax-errors=5.
*** [.pio\build\STM32H743VI_btt\src\src\HAL\STM32\HardwareSerial.cpp.o]
Error 1
Marlin\src\HAL\STM32\../../inc/../core/serial.h:56:26: error: 'MYSERIAL1'
was not declared in this scope; did you mean 'MYSERIAL2'?
   56 |   #define _SERIAL_LEAF_1 MYSERIAL1

Marlin\src\HAL\STM32\dogm\../../../inc/../core/serial.h:56:26: error:
'MYSERIAL1' was not declared in this scope; did you mean 'MYSERIAL2'?
   56 |   #define _SERIAL_LEAF_1 MYSERIAL1

*** [.pio\build\STM32H743VI_btt\src\src\HAL\STM32\HAL.cpp.o] Error 1

*** [.pio\build\STM32H743VI_btt\src\src\HAL\STM32\Servo.cpp.o] Error 1
In file included from
Marlin\src\HAL\STM32\dogm\../../../inc/MarlinConfig.h:41,
                 from
Marlin\src\HAL\STM32\dogm\u8g_com_stm32duino_swspi.cpp:25:
Marlin\src\HAL\STM32\dogm\../../../inc/../core/serial.h:109:56: error:
template argument 1 is invalid
  109 |   typedef MultiSerial< REPEAT_1(NUM_SERIAL, _S_MULTI) 0>
SerialOutputT;
      |                                                        ^
Marlin\src\HAL\STM32\dogm\../../../inc/../core/serial.h: In function 'void
SERIAL_CHAR(char, Args ...)':
Marlin\src\HAL\STM32\dogm\../../../inc/../core/serial.h:132:55: error:
request for member 'write' in 'multiSerial', which is of non-class type
'SerialOutputT' {aka 'int'}
  132 | void SERIAL_CHAR(char a, Args ... args) { SERIAL_IMPL.write(a);
SERIAL_CHAR(args ...); }
      |                                                       ^~~~~
compilation terminated due to -fmax-errors=5.
***
[.pio\build\STM32H743VI_btt\src\src\HAL\STM32\dogm\u8g_com_stm32duino_swspi.cpp.o]
Error 1
In file included from Marlin\src\HAL\STM32\../../inc/../HAL/HAL.h:30,
                 from
Marlin\src\HAL\STM32\../../inc/MarlinConfigPre-5-post.h:29,
                 from
Marlin\src\HAL\STM32\../../inc/MarlinConfigPre-6-type.h:24,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\STM32\eeprom_flash.cpp:26:
Marlin\src\HAL\STM32\../../inc/../HAL/../HAL/STM32/HAL.h:66:4: error:
#error "SERIAL_PORT must be from 1 to 9, or -1 for Native USB."
   66 |   #error "SERIAL_PORT must be from 1 to 9, or -1 for Native USB."
      |    ^~~~~
In file included from
Marlin\src\HAL\STM32\../../inc/MarlinConfigPre-1-axes.h:40,
                 from
Marlin\src\HAL\STM32\../../inc/Conditionals-1-axes.h:32,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfigPre.h:27,
                 from
Marlin\src\HAL\STM32\../../inc/MarlinConfigPre-5-post.h:24,
                 from
Marlin\src\HAL\STM32\../../inc/MarlinConfigPre-6-type.h:24,
                 from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:28,
                 from Marlin\src\HAL\STM32\eeprom_bl24cxx.cpp:32:
Marlin\src\HAL\STM32\../../inc/../core/serial.h:56:26: error: 'MYSERIAL1'
was not declared in this scope; did you mean 'MYSERIAL2'?
[... 127 lines cut ...]
      |                        ^~~~~
Marlin\src\HAL\STM32\../../inc/../core/macros.h:642:30: note: in expansion
of macro 'EVAL1'
  642 | #define EVAL2(V...)    EVAL1(EVAL1(V))
      |                              ^~~~~
Marlin\src\HAL\STM32\../../inc/../core/macros.h:641:30: note: in expansion
of macro 'EVAL2'
  641 | #define EVAL4(V...)    EVAL2(EVAL2(V))
      |                              ^~~~~
Marlin\src\HAL\STM32\../../inc/../core/macros.h:640:30: note: in expansion
of macro 'EVAL4'
  640 | #define EVAL8(V...)    EVAL4(EVAL4(V))
      |                              ^~~~~
Marlin\src\HAL\STM32\../../inc/../core/macros.h:639:24: note: in expansion
of macro 'EVAL8'
  639 | #define EVAL16(V...)   EVAL8(EVAL8(V))

*This is Marlin 2.1.3-b2*
If I am going to go through all of this, I am going to do it on the latest
and greatest.

When all is done and working, I will happily share my Configuration.h and
any other files you would like to put in your library.

Bob Sisson

inspectionsbybob, 2025-02-02:
Do I need serial port 3, or is that for the Pi later?

The Marlin documentation says  -1 to 9.
If you decide to release a 2.1.3.-b2 config you may want to make some notes
in the serial section about specific needed for your board.

 Select the serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to
non-default port pins.
 * Serial port -1 is the USB emulated serial port, if available.
 * Note: The first serial port (-1 or 0) will always be used by the Arduino
bootloader.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7, 8, 9]
 */
#define SERIAL_PORT 1

Bob Sisson

inspectionsbybob, 2025-02-02:
SUCCESS!   Lots of Yellow warnings and comments, but no errors. Mostly
about the Z_min_probe.

I will move this build to the printer later (I have some cable neatening to
do; I undid some stuff while waiting.)

I have the SKR3ez, an external MOSFET for the Bed, and a spare MosFet for
the Nozzle if I ever upgrade that.
The Supply is a 500W I had in another printer that was brain-dead.
I put in a separate 24V rail for Fans, lights, and other things, as that is
easier than using the NO connections. The Case has an 80mm that sits above
the 2209's, and the PS has its own that is always on (noisy, that's why
everything is in the Back)

Once stable, the TFT35 will be moved to the front. Currently, it is
connected through a 5-pin extension cable.

The Case didn't align with the USB port, so I have a short USB-B pigtail
that brings that outside the Case.

Getting into the MicroSD card is tight but doable. Since I don't use it for
printing, it is okay.

[image: IMG_9964[1].JPG]

Bob Sisson

inspectionsbybob, 2025-02-02:
I Moved the build to the Printer...
It did grab the Firmware.bin file...

Neither the USB port or the TFT35 responded.

The TFT35 said "Marlin Mode" and nothing else.
The USB Port didn't register to a connected PC.
I checked the USB/CAN switch.

I am using the TFT35 as it came, I have not flashed anything "new" assuming
that the default firmware would work fine for Marlin Mode.

I am more worried about the USB port not working, as I use that for 90% of
the time.

I have been using (or trying to ) the STM32H743VI_btt.  Should I be using
the STM32H743VG_btt build option?

So what to try next?

Bob Sisson

inspectionsbybob, 2025-02-02:
Here is my current configuration.h

I am sure there is a typo somewhere, but I can't see it...

[Configuration.zip](https://github.com/user-attachments/files/18633646/Configuration.zip)

rondlh, 2025-02-03:
Marlin mode will not work when you only connect the 5-pin connector,  then only touch mode will work. 
The TFT needs to be set to the same baudrate as the SKR 3, which is set to 115200 baud.

You should see an extra serial port when you connect the motherboard to a PC,

An update of the TFT is definitely recommended.

inspectionsbybob, 2025-02-03:
Thanks...
I was mis-informed about what mode was supported by the 5-pin as I was
originally told that it (the 5-pin) was the Marlin mode and I could extend
that as needed.  I will pull the latest build for the TFT from git hub for
V3.  Wish the TFT had a "setup" mode where you could change the baud rate.

-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

rondlh, 2025-02-04:
> Wish the TFT had a "setup" mode where you could change the baud rate.

What else do you want to setup in the "setup mode"?

A auto detect of the baudrate would be very useful.

inspectionsbybob, 2025-02-04:
Anything that prevents the TFT from doing "something".  A blank screen does
nobody any good. Error msgs are really really useful.

Maybe when at the marlin/touch choice screen, a tripple tap would bring up
a service menu.

I realize that I enabled M575, so I can cycle through the rates, but that
will be a pain and I don't know if I would need to reset the tft between
tries...

-----------------------------------------------------------------------------------
Need a quick answer? Call our booking line at (301) 208-8289, or book your
inspection on our website 24/7.
Don't wait for a response as someone else may get the spot you want... call
Today!
-----------------------------------------------------------------------------------
Bob Sisson, ACI, BVI
Inspections by Bob, LLC
www.inspectionsbybob.com
***@***.***
MD Lic#29666; ASHI Member #212016

rondlh, 2025-02-04:
> Anything that prevents the TFT from doing "something". A blank screen does nobody any good. Error msgs are really really useful.

In touch mode, you will never have a blank screen.

> Maybe when at the marlin/touch choice screen, a tripple tap would bring up a service menu.

I'm afraid the LLM in the TFT currently does not support triple tapping.

> I realize that I enabled M575, so I can cycle through the rates, but that will be a pain and I don't know if I would need to reset the tft between tries...

You can change the TFT baudrate without resetting. The TFT will continue to try to establish a connection.

inspectionsbybob, 2025-02-05:
THANKS !

That info is useful

Bob Sisson