bigtreetech/BIGTREETECH-TouchScreenFirmware #2972: [Q] Issues TFT70 V3.0 combined with BTT Manta M5P with CB1

likes3Dprinting, 2025-02-11:
Hi All,

I have a BTT Manta M5P board equipped with a CB1 module. I also have a TFT70 V3.0 display that I'm trying to use. I'm having significant troubles getting the board to communicate with the display. It won't receive communication from the Manta board and keeps reporting 'No printer attached!' while in 'Touch Mode'.

The Manta M5P board comes without a dedicated TFT connector and thus I'm trying to get it to communicate through UART. To no avail yet...

From issue #2878 I read that the manufacturer (somehow?) doesn't expect one to attach the TFT70 V3.0 to a Manta M5P board. However, in my understanding it should be possible to get it to communicate over UART. 

In the printer.cfg I've added a [display] section where I've tried several options under 'lcd_type:'
Here, 'serial' and 'uart' are not valid options, st7920 and tft35 are both SPI displays and as such that won't work either. 
```
[display]
lcd_type: #serial #uart #st7920
serial_port: /dev/ttyS0  # or the appropriate serial port
baud_rate: 115200
tx_pin: GPIO14  # UART TX pin
rx_pin: GPIO15  # UART RX pin
```

From the Manta M5P the UART0 is enabled, communicating at 115200 baudrate. When using `sudo cat /dev/ttyS0` this returns `M105`. Which is a Temperature Request in G-code, indicating that the display is likely trying to communicate with the printer firmware. That is, communication seems to happen...

How can I configure the Manta M5P board such that it communicates over UART using GPIO14 and GPIO15 to the TFT70 board? Does anyone have ideas/could give me some advise?

rondlh, 2025-02-11:
The motherboards seems to be based on a STM32G0B1, which does not have a UART0.

On EXP2 you have:
PB8     USART3 TX
PB9     USART3 RX

radek8, 2025-04-17:
What software do you have on your Manta board?