bigtreetech/BIGTREETECH-TouchScreenFirmware #2974: Flash MKS GD TFT28

AleSan78, 2025-04-12:
Hi, when I compile the firmware for a MKS GD TFT28 screen, the screen flash stops at "complete" and remains blocked, what should I do, thank you

rondlh, 2025-04-15:
> Hi, when I compile the firmware for MKS GD TFT28 screen, the screen flash stops at "complete" and remains blocked, what should I do, thank you. I specify the screen version is GD32F305 with bootloader v3.0.5

Try this:
Create a file named "reset.txt" on a SD card, insert it into the TFT, then power on the TFT. This procedure resets all TFT settings to their safe startup defaults.

AleSan78, 2025-04-16:
No, that's not what I'm asking, I can flash Kisslorand's firmware, I'd like to flash my own firmware, but it gets stuck at the "complete" text.

rondlh, 2025-04-17:
Better stay clear of Kisslorand's firmware, it's not peer reviewed and known to be slow and buggy. It can damage both your printer and TFT.

AleSan78, 2025-04-17:
It may be buggy but it's the only one I can flash

AleSan78, 2025-04-17:
When I try your method with "reset.txt", I restart the printer when the text "complete" is displayed and remains stuck, I press the reset button on the screen then it remains stuck on "v3.0.5"

radek8, 2025-04-17:
I'm not sure if your display version is supported by this FW.

kisslorand, 2025-04-17:
> Better stay clear of Kisslorand's firmware, it's not peer reviewed and known to be slow and buggy. It can damage both your printer and TFT.

LOL, if you say so...

Thanks for the unsolicited review thou. Funny how the loudest critics are always the ones with zero facts and plenty of noise. Next time, try including a single verifiable fact — it might help your credibility. Otherwise, stay in your lane. Quality work isn’t built on anonymous whining.
Now be a good little boy, go back to your room, and play nicely with your toys. Leave the real work to the grown-ups.

AleSan78, 2025-04-17:
Peace and love guys, I just want to compile my firmware and figure out why it's not working. I'm using Kisslorand's and I don't see any problems so far.

rondlh, 2025-04-25:
> Peace and love guys, I just want to compile my firmware and figure out why it's not working. I'm using Kisslorand's and I don't see any problems so far.

If you compile for the correct target hardware then flashing should work. I guess the flashing actually works, just the firmware doesn't start. This could be because there is a conflict with the settings preventing startup, to solve this do the "reset.txt" thing mentioned above. 
Or perhaps the display is in Marlin mode? To change the TFT mode press the encoder button for a few seconds.

Does flashing of the official firmware provided here work?

AleSan78, 2025-04-27:
> If you compile for the correct target hardware then flashing should work. I guess the flashing actually works, just the firmware doesn't start. This could be because there is a conflict with the [...]

No, I've already told you that the flash doesn't finish and gets stuck on "complete" when it should be flashing the fonts and BMP icons.
However, when I compile Ciotto's modified firmware, it works.

rondlh, 2025-04-28:
You can use Ciotto's firmware without worries, it's a tested and peer reviewed.

TimoSEL, 2025-07-01:
I have the same Problem as AleSan78. Stuck after flash at "complete" and after reset only shows "3.0.5".
Hardware Sidwinder X2 with Display Hardware-Version 1.2 (GD32F305) GD32F305 with bootloader v3.0.5