bigtreetech/BIGTREETECH-TouchScreenFirmware #2976: [BUG] Print head crashes when exiting "Leveling" menu

hafunui, 2025-04-29:
### Description

When I enter the "Leveling" menu and then back out of that menu, the print head starts moving down until it crashes. I'm trying to configure the newest version of marlin on my custom delta printer (modified anycubic kossel linear plus), so I'm not completely sure everything is configured properly yet. I did have a manually deployable probe attached in anticipation for auto bed leveling. In my old configuration, with marlin-bugfix-2.0.x and stock reprap discount smart controller, it would prompt me to deploy a probe, regardless if one was already attached, and await my confirmation. No such prompted occurs here, so I'm not sure what command the TFT sent to the mainboard. It just moves down and crashes. (And destroyed my probe)

The "Bed Leveling" menu doesn't do this. It seems to work fine.

### Steps to reproduce

1. Navigate to "Movement > Leveling" menu on TFT
2. Press Back
3. Print head moves down until crash and stall.

**Expected behavior**
Nothing. Backing out of menu shouldn't move any motor.

**Actual behavior**
Print head moves.

### Hardware Variant
TFT35 V3.0.1
BIGTREETECH-SKR-PRO-V1.1

### TFT Firmware Version & Main Board Firmware details
BIGTREE_GD_TFT35_V3.0.27.x
Marlin-2.1.3-b2

### Additional Information
Attached is a zip containing TFT35 config.ini and marlin configurations, if it helps.
[TFT35_Marlin_configs.zip](https://github.com/user-attachments/files/19952870/TFT35_Marlin_configs.zip)

hafunui, 2025-05-04:
So it turns out the crashing was due to an incorrect `DELTA_HEIGHT` in marlin. With a correct height configured, the printhead now parks a safe distance away from the bed. 

However, I'm still not sure this is intended behavior. Simply entering and exiting the leveling menu could cause the head to crash into partially printed or cooling parts, for example.

rondlh, 2025-05-05:
I had a quick look into this issue. 
What happens is this:
In Leveling.c there is this code for when you exit the Leveling menu:
```
      case KEY_ICON_7:
        probeHeightMove(infoSettings.level_z_raise);

        CLOSE_MENU();
        break;
```
probeHeightMove moves the nozzle away from the bed by "level_z_raise" as defined in the config.ini file.
This seems to there for safety, like for example when you manually leveled the bed by letting the nozzle touch the bed, but never moved the nozzle away from the bed afterwards, which could scratch the bed if you make a move in this condition.
So in general this seems to be a good thing to do.

closed 2025-06-30 by hafunui