bigtreetech/BIGTREETECH-TouchScreenFirmware #2977: Adding SD(mSD) reader to tft35 v2

slayernn021, 2025-05-02:
Hello. I have tft35 v2 and built SD reader is not suitable for panel i'm making as SD slot needs to be in front near LCD. How can i add simple arduino SD/Micro SD reader without messing too much (unsoldering stock SD etc etc), is there a way to use any of ports on tft35 for this use and how? Tnx in advance

rondlh, 2025-05-06:
Unfortunately the TFT35 V2 does not have a build in external SD card port to relocate the SD card slot.
There are SD Card extender cables, one end is like a SD card, that you just insert into the TFT SD Card slot, the other end is a SD card slot that you can mount were needed.

slayernn021, 2025-05-06:
Tnx man. I was happy with all awailable ports on tft and later discovered its useless (like wifi,still not developed etc). Im printing enclosure for lcd tilted 8deg (still looks slim) so sd can be easily removed from bottom. Will not go exender way

rondlh, 2025-05-06:
> Tnx man. I was happy with all awailable ports on tft and later discovered its useless (like wifi,still not developed etc). Im printing enclosure for lcd tilted 8deg (still looks slim) so sd can be easily removed from bottom. Will not go exender way

I'm not sure if I understand your SD card slot situation, a picture might clarify a lot.

About Wifi support.
The TFT can work well with an ESP Wifi module running ESP3D (2.1/3.0). Are there any specific features you are looking for?

rondlh, 2025-05-07:
Thanks for the clarification, I see what you are trying to do.
The BTT TFT35 V2.0 does not have a wifi port or free serial port to connect a ESP wifi module to.

On supported hardware, wifi is supported through ESP3D. I use it on all my printers to monitor the printer status and issue Gcode commands or start a print job remotely (file is at the TFT).

closed 2025-07-14 by github-actions[bot]