bigtreetech/BIGTREETECH-TouchScreenFirmware #2979: [Q] (Your Question?)BL Touch setup issues with Marlin 2.1.2.4 for SKR Mini E3 V3.0 STM32G0B1RE with BLTOUCH for Z homing

ldswofford, 2025-05-22:
**Your question or details you need. Please describe.**
trying to setup a new BTT SKR Mini E3 V3.0 and BL Touch (3D Touch) on an older Ender-3 S1.

**specify the device variant related to your question.**

**Additional context**
Add any other context or screenshots about the question here.
I am somewhat new to 3D printing/updating/marlin/BTT/bltouch/etc.
 I have updated an Ender-3 S1 with the BTT SKR Mini E3 V3 board. I have that working (somewhat using the initial Bug-Fix firmware that came with the board. I have now added a BL Touch (3D Touch) to the board, using the board's BL Touch connector(s), not the z-stop connector/with bltouch pins.
 I have been unable to get the original canned firmware from Github for the ender-3 bltouch to work correctly (won't move to center on autohome), so I updated to this Marlin 2.1.2.4 for SKR MINI E3 V3.0, etc...that has the z-homing... The issue is that when autohoming, the probe test doesn't stop and ends up ihitting on the bed and making 'the bed noise'. I have made changes to the -1.85 z-stop setting in the config as far as up to +2.5 and the probe still never stops the nozzle from hitting the bed. (yes, I test/save/reboot/etc and have reloaded the firmware to reset everything).The comment above from in the firmware set discussion from bcat, makes it sound like no changes are needed and this firmware will work for both using z-stop, or using the board's bltouch 5 pin port for everything, leaving the z-stop connection empty... is this true, or do I have to modify the 3d touch plugs to create one to install in the z-stop connector? Or... do I have to modify the firmware options to disable the z-stop connector? Thanks

rondlh, 2025-05-22:
Sounds like a Marlin configuration issue to me, not like a BTT Touchscreen firmware issue.

ldswofford, 2025-05-22:
rondlh2, this is about BL Touch... not touchscreen. It could be a configuration issue, but I am using what is supposed to be a 'canned' firmware bin that has been out for around 6 months to a year on github. I am not brave enough to believe I am ready to tackle attempting my own custom firmware mods yet.... :-)

rondlh, 2025-05-24:
> won't move to center on autohome

This Github page here is for the BTT touchscreen firmware, which can send an auto home command (G28), but the execution is up to Marlin (or RRF). If the probe doesn't go to the center of the bed, then perhaps you have a firmware for another bed size or your steps per unit (M92) doesn't match your printer.
You can run M115 (if supported) to get more insight into your bed size and offsets, check the "area" reported.

closed 2025-07-31 by github-actions[bot]