bigtreetech/BIGTREETECH-TouchScreenFirmware #2982: [BUG] Display not updating temperatures before print starts

dmarnel, 2025-07-07:
### Description

Using OctoPrint on a Raspberry Pi connected to my Ender 3 Pro, the temperatures are not updating on my BTT display until the print begins. However, I do see the echoed messages saying "Bed heating...", and "E1 heating...", etc.

I am using the latest Marlin and BTT firmware as of 2025-07-07

Machine: Creality Ender 3 Pro
Printer board: BTT SKR Mini E3 V2.0
	chip: STM32F103 RCT6
Display board: BTT TFT35 E3 V3.0
	chip: STM32F207 VCT6

### Steps to reproduce

1. Access OctoPrint from the browser
2. start a print

**Expected behavior**
As the extruder and bed heat up, the display should update the temperatures.

**Actual behavior**
The temperatures are only updated on the display once the print begins.

### Hardware Variant
Display board: BTT TFT35 E3 V3.0, chip: STM32F207 VCT6

### TFT Firmware Version & Main Board Firmware details
BTT: BIGTREE_TFT35_V3.0_E3.27.x.bin
Marlin: bugfix-2.1.x

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

[Configuration.h.zip](https://github.com/user-attachments/files/21110375/Configuration.h.zip)

rondlh, 2025-07-09:
I never came across an issue like this, but anyway.
Don't use the firmware in "Releases", it's a firmware from 2021 which is quite outdated.
All version after that have the same version number (.27), so you must use the firmware date to confirm you are using the latest firmware.
Download the firmware from the directory: "Copy to SD Card root directory to update".
Currently the latest firmware is from May 1st 2025.

dmarnel, 2025-07-09:
Hi rondlh, thank you for the response. I'm not sure what you mean by 'Don't use the firmware in "Releases"'.

I cloned the repo at https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware, and am using the files from the master branch. When I view the BTT: BIGTREE_TFT35_V3.0_E3.27.x.bin on github it says it's last commit date was 3 months ago, so I assume that is the binary I should be using?

rondlh, 2025-07-11:
> Hi rondlh, thank you for the response. I'm not sure what you mean by 'Don't use the firmware in "Releases"'.
>
> I cloned the repo at https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware, and [...]

Yes, I think you are using the correct/latest version. On Github you can also find "Releases", on the right side of the main page. But for the BTT Touchscreen firmware the last release is completely outdated. As you issue seems to be quite odd, I worried that you might be using a very old version. I'm not sure what is causing your issue.

The temperature data on the TFT is picked up from the Marlin temperature reporting, which is controlled by Gcode M105/M155.
A valid temperature is required for the TFT to know that it is connected to a motherboard.
If you go to the TFT Terminal screen, and press the "OFF" button (should change to "ON"), then  you should be able to see the temperature reports on the TFT. Please have a look...

dmarnel, 2025-07-14:
I don't see an OFF button on the Terminal screen.

A couple of other things I have noticed:

If I go to the temperature tab in OctoPrint and increase the temperatures, I do see the temperatures being updated on the BTT touchscreen.

When I use the (un)load button on the touchscreen to load filament and then press CONTINUE when loading is complete, if I do not navigate anywhere else after pressing CONTINUE and I start a print from OctoPrint, when I press the back arrow on the touchscreen I get a popup window saying something like "Wait for the loading process to complete." I can't navigate anywhere else using the touchscreen because it thinks the printer is still loading filament. Once printing begins, that popup window no longer appears and I can press the back arrow to return to the main screen.

It seems like something is blocking the BTT functionality when a print is started from OctoPrint, while the nozzle/bed are heating. Once heating is complete and the print begins everything works again.

rondlh, 2025-07-22:
> ￼
> 
> ￼
> […](#)
 

I don't understand what this is, could you please clarify?

closed 2025-09-28 by github-actions[bot]