bigtreetech/BIGTREETECH-TouchScreenFirmware #2986: [Q] Can't install any version on Artillery X2

xCISACx, 2025-08-24:
I can't install either version from this repository on my Artillery Sidewinder X2.

My Board is MKS_GD_TFT28_V4.0 according to my current TFT firmware.
The only firmware I have managed to install is V27.x from March 28 2023, provided by JC3Design in his [video](https://youtu.be/1xefD6rvddI?si=DXsq5wiBL6CIIKRO)

I thought the issue might have been an outdated Marlin firmware since I installed 2.1.2.1 according to his video, so I updated Marlin to 2.1
3-beta3 and it's working fine with his TFT version.

My problem with this version is that the information popups at the top are in Spanish and I'd like to change them to English.

Every time I tried to install a firmware from this repository, I got blocks instead of fonts, an unresponsive screen or sometimes even a combination of both.

Which firmware should I actually be using for my printer?

Here is a picture of my current info screen:

[image]

Any help would be greatly appreciated 🙂

Thanks!

kisslorand, 2025-08-31:
You can try [this](https://github.com/kisslorand/BTT-TFT-FW).
Most probably some troll will jump in to tell you that it will burn your house or whatnot but just ignore him and try it. You have nothing to loose.

rondlh, 2025-09-06:
@xCISACx Be careful with kisslorand's closed source firmware, it's unreliable, buggy and slow. His firmware is not peer reviewed and not supported here, it might damage your TFT and/or printer. But the chance of it burning down your house is quite small.