bigtreetech/BIGTREETECH-TouchScreenFirmware #2989: [BUG] TFT35 B1 v3.0.1 cannot send commands except after reconnection

EmmanuelMess, 2025-09-10:
### Description

We have a Fly-E3-Ultra Max H723 board with RRF 3.6.0.
We bought a BTT TFT35 B1 v3.0.1 screen, and connected it with EXP1 and EXP2 to provide GND and +5V, and Tx and Rx. The screen was flashed with binary BIGTREE_GD_TFT35_V3.0_B1.28.x.bin, the instruction "M575 P1 S2 B57600" was added to config.g and it seemed to work. The issue is that values that should update on the screen, don't update (for example temperature remains at the first read value) and when I press home all, the printer doesn't move. But if I go to settings > connection > disconnect and then reconnect, it executes all previous instructions.

I also tried connecting only the cables for the "screen" connector (no EXP1 or EXP2), same result as before. Also tried to switch the Tx and Rx cables, but this makes the printer not be recognized by the screen.

RRF 3.6.0
TFT: BIGTREE_GD_TFT35_V3.0_B1.28.x.bin

### Steps to reproduce

1. Flash screen with BIGTREE_GD_TFT35_V3.0_B1.28.x.bin
2. Connect via serial to Fly-E3-Ultra Max H723
3. Send home all via UI
5. Disconect via the button in the settings
6. Reconnect via touching the screen

**Expected behavior**

1. Flash screen with BIGTREE_GD_TFT35_V3.0_B1.28.x.bin
2. Connect via serial to Fly-E3-Ultra Max H723
3. Send home all via UI
4. Homes all axis

**Actual behavior**
1. Flash screen with BIGTREE_GD_TFT35_V3.0_B1.28.x.bin
2. Connect via serial to Fly-E3-Ultra Max H723
3. Send home all via UI
4. Nothing happens
5. Disconect via the button in the settings
6. Reconnect via touching the screen
7. It homes all axis

### Hardware Variant
 BIGTREE_GD_TFT35_V3.0_B1.28.x.bin
### TFT Firmware Version & Main Board Firmware details
We have a Fly-E3-Ultra Max H723 board with RRF 3.6.0.

### Additional Information

Images are from the "screen" connector only.

[image]

[image]

[image]

EmmanuelMess, 2025-11-10:
This is still a problem.

kisslorand, 2026-09-12:
This repository is dead, it is also known it is not RRF friendly, even if they stated so.
You can instead do what also TeamGloomy officially recommends, use the FW from my repository: https://github.com/kisslorand/BTT-TFT-FW.
Reference: https://teamgloomy.github.io/tft.html

> [!NOTE]
> If user @rondlh comes warning you that the mentioned firmware will eat your cat or whatever other bull$hit, just ignore him, he's just a goofy BTT fanboy.

P.S.: Found this randomly: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/3019, check how easy was the fix. :)

kisslorand, 2026-09-12:
[image]

kisslorand, 2026-09-12:
[image]

EmmanuelMess, 2026-09-12:
> This repository is dead, it is also known it is not RRF friendly, even if they stated so. You can instead do what also TeamGloomy officially recommends, use the FW from my repository: [kisslorand/BTT-TFT-FW](https://github.com/kisslorand/BTT-TFT-FW?rgh-link-date=2026-09-12T20%3A51%3A12.000Z). Reference: [teamgloomy.github.io/tft.html](https://teamgloomy.github.io/tft.html)
> 
> Note
> 
> If user [@rondlh](https://github.com/rondlh) comes warning you that the mentioned firmware will eat your cat or whatever other bull$hit, just ignore him, he's just a goofy BTT fanboy.
> 
> P.S.: Found this randomly: [#3019](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/3019), check how easy was the fix. :)

Your repo has the same issue, https://github.com/kisslorand/BTT-TFT-FW/issues/30 . And you stopped working on it a bit after I got the screen. I think this issue is specific to my model of screen though.

kisslorand, 2026-09-13:
I am sorry to hear that but since now I only have Klipper machines I cannot do anything to fix any issue or inconvenience.

EmmanuelMess, 2026-09-13:
> I am sorry to hear that but since now I only have Klipper machines I cannot do anything to fix any issue or inconvenience.

No worries, thanks for the work on the fork though.